femtofox/femtofox #20: Todo: Test power consumption with LoRa radio attached / figure out what size solar panel will be required

noon92, 2024-12-06:
Preliminarily, 0.36-0.45w average

joshbowyer, 2025-01-13:
My test with the Luckfox Pico Mini, with wifi on via AR9271, and an open ssh session, running off 3x 18650 (6600mAh) batteries through a 5vout MPPT, is precisely 48 hours of runtime. This is about .69Wh, so given solar peak hours in the northern hemisphere (particularly northern USA), We would be able to run this off a 5W panel with decent solar hours according to this explanation: https://docs.google.com/document/d/12GIY24vLKLABg2RUTPP6yMzokr44GMzJOE4p7ngaCbI/edit?tab=t.0#heading=h.9lmvuqjahqxl

So realistically for more northerly climates, a minimum of a 10W panel should be considered. Southerly users may be able to comfortably use a 5W panel.

noon92, 2025-01-13:
This matches my expectations.

mikecarper, 2025-04-12:
benchtop power supply is using roughly 0.4w-0.7w for the computer. Add on the lora radio for more power usage of 3w