femtofox/femtofox #26: Todo: allow meshtastic wifi config to change wifi ssid and psk

noon92, 2024-12-10:


noon92, 2024-12-11:
@Ruledo please note

To allow meshtastic and usbconfig to change wifi settings while still allowing user to set multiple SSIDs with different priorities in wpa_supplicant.conf, we will have a primary SSID with a priority of 100. All other SSIDs will have lower priorities. Meshtastic and USBconfig will only edit the `priority: 100` SSID.

noon92, 2024-12-13:
This is more annoying than I thought. Without the ability to change the setting in meshtastic's config.proto, meshtastic settings will always overwrite wpa_supplicant.conf. So we need a way to write to config.proto. Also, apparently every time I access config.proto it causes meshtastic to crash (what?), so instead of directly examining the file, we'll just monitor the last modified time, and then check the values only if that changes.

So we'll also need to monitor wpa_supplicant.conf for changes, and if THAT changes, update config.proto. What a headache.

```#!/bin/bash
ip link set wlan0 up
#sleep 60 # wait 60 seconds so networking will be available during boot

WPA_SUPPLICANT_CONF="/etc/wpa_supplicant/wpa_supplicant.conf"
CONFIG_PROTO="/root/.portduino/default/prefs/config.proto"

wifichanged="false"

# Initial value for comparison (start with impossible values so they'll always implement the current values after boot)
previous_value_4_1=2

previous_value_4_3=$(awk '/network=/ {in_network=1} /}/ {in_network=0} in_network && /priority=100/ {print} in_network' "$WPA_SUPPLICANT_CONF" | grep -oP 'ssid="\K[^"]+')
previous_value_4_4=$(awk '/network=/ {in_network=1} /}/ {in_network=0} in_network && /priority=100/ {print} in_network' "$WPA_SUPPLICANT_CONF" | grep -oP 'psk="\K[^"]+')

# Get the initial modification time of the config.proto file
previous_mod_time=$(stat -c %Y "$CONFIG_PROTO")

# Flag to ensure the first iteration runs
first_run=true

while true; do
  current_mod_time=$(stat -c %Y "$CONFIG_PROTO")

  if $first_run || [ "$current_mod_time" != "$previous_mod_time" ]; then
    if systemctl is-active --quiet meshtasticd; then
      # Run the command to fetch the values
      current_value_4_1=$(cat "$CONFIG_PROTO" | protoc --decode_raw | awk '/4 {/, /}/ {if ($1 == "1:") print $2}')
      current_value_4_3=$(sudo cat "$CONFIG_PROTO" | protoc --decode_raw | awk '/4 {/, /}/ {if ($1 == "3:") print substr($0, index($0,$2))}')
      current_value_4_4=$(sudo cat "$CONFIG_PROTO" | protoc --decode_raw | awk '/4 {/, /}/ {if ($1 == "4:") print substr($0, index($0,$2))}')

      # Check if the value for 4:1 has changed
      if [ "$current_value_4_1" != "$previous_value_4_1" ]; then
        if [ "$previous_value_4_1" -eq 1 ]; then
          prev_text="Wifi on"
        elif [ "$previous_value_4_1" -eq 2 ]; then
          prev_text="Startup"
        else
          prev_text="Wifi off"
        fi
        if [ "$current_value_4_1" -eq 1 ]; then
          cur_text="Wifi on"
        else
          cur_text="Wifi off"
        fi

        # Log the change
        echo "Wifi mesh control: wifi setting changed: $prev_text -> $cur_text"

        # Update the previous value for next comparison
        previous_value_4_1="$current_value_4_1"
      fi

      # Check if the value for 4:3 has changed
      if [ "$current_value_4_3" != "$previous_value_4_3" ]; then
        echo "Wifi mesh control: 4:3 setting changed: $previous_value_4_3 -> $current_value_4_3"
        sed -i '/priority=100/,/}/s@ssid="[^"]*"@ssid='"$current_value_4_3"'@g' "$WPA_SUPPLICANT_CONF"
        previous_value_4_3="$current_value_4_3"
        wifichanged="true"
      fi

      # Check if the value for 4:4 has changed 
      if [ "$current_value_4_4" != "$previous_value_4_4" ]; then
        echo "Wifi mesh control: 4:4 setting changed: $previous_value_4_4 -> $current_value_4_4"
        sed -i '/priority=100/,/}/s@psk="[^"]*"@psk='"$current_value_4_4"'@g' "$WPA_SUPPLICANT_CONF"
        previous_value_4_4="$current_value_4_4"
        wifichanged="true"
      fi

      if [ "$wifichanged" = "true" ]; then
        sudo systemctl restart wpa_supplicant
        sudo wpa_cli -i wlan0 reconfigure
        sudo dhclient
        wifichanged="false"
      fi

      # Take action based on the new value of 4:1
      if [ "$current_value_4_1" == "1" ]; then
        ip link set wlan0 up
      else
        ip link set wlan0 down
      fi

    else
      if ip link show wlan0 | grep -q 'state DOWN'; then
        ip link set wlan0 up
        logger "Wifi mesh control: Meshtasticd service is offline, turning wifi on."
      fi
    fi

    # Update the modification time for the next iteration
    previous_mod_time="$current_mod_time"
    # Set the flag to false after the first run
    first_run=false
  fi

  # wait 15 secs before checking again
  sleep 15
done
```

Ruledo, 2024-12-14:
Current wifi-mesh-control service shuts off wifi after a few moments after boot.
Disabling the service and wifi works fine.
Putting this here because I know it's related.
sudo systemctl disable wifi-mesh-control   for now.