femtofox/femtofox #90: Todo: Before move to 1.0, minimize/remove logging to reduce SD card wear

noon92, 2024-12-28:
@SpudGunMan

noon92, 2024-12-28:
Both system logging and meshtastic logging.

joshbowyer, 2025-01-13:
Something we could do is make /var/log immutable, so that nothing gets written to disk but journalct will still show all logs for the current session

SpudGunMan, 2025-01-13:
Just saw this yea that would be handy. For meshbot it can go no or little logs but having something would be nice

noon92, 2025-01-13:
> Something we could do is make /var/log immutable, so that nothing gets written to disk but journalct will still show all logs for the current session

A system-wide "turn off logs" switch would be really great, so I like this a lot. You say journalctl will still show logs for current boot - I assume they're stored in ram. We don't have much ram, so long uptimes will probably exceed logging capacity. I assume the OS will dump the logs in this case?

PR #206 (2025-01-14): Add "disable logging" feature
resolves #90 

@joshbowyer
closed 2025-01-14 by noon92