femtofox/femtofox #225: Remove rc.local and luckfox-config on boot

joshbowyer, 2025-01-20:
This didnt make any discernible impact on boot for me in my testing, aside from faster boot. Disabling rc-local.service should be sufficient.

Ruledo, 2025-01-20:
Actually my new code uses and override to start meshtasticd after rc-local so requires the service.
Also, I think one of nagu's scripts might inject something into it.  Need to check.  But I wanna leave this open so we can try.  Will check femto-config soon but if I remember right it was in stock rc.local.  would be way cleaner if we can pull this off though.

Ruledo, 2025-01-20:
Luckfox-config needs to be loaded.
Otherwise you can't access gpio in bash and other things.

noon92, 2025-02-03:
It appears luckfox-config load isn't needed. rc.local is still used for a couple things.

noon92, 2025-02-11:
Luckfox config is gone. a couple things are left in rc.local that don't have a more appropriate home.

closed 2025-02-11 by noon92