femtofox/femtofox #326: Contact Client not displaying UI when choosing Run Software within femto-config

pdxlocations, 2025-02-13:
Tiny bug, not sure if it's a Foxbuntu thing or a Contact Client thing.  SFC -> Software Manager -> Contact -> Run Software.  The client does launch (contact log populates) but is not displayed on screen.  My guess is curses somehow fails to take over the window with FC running.  Easy fix would be to remove the option to run from within the dialogue and/or exit config before running.

noon92, 2025-02-14:
Yeah you're right. Hmmm I don't know what's causing this

noon92, 2025-02-14:
The command is being sent correctly. This feels like it's something in contact, but I can't imagine what. It's very odd.

noon92, 2025-02-14:
Temporarily have removed the run software option from menu.

noon92, 2025-03-08:
Closed by #386

closed 2025-03-08 by noon92