femtofox/femtofox #355: femto_mac_address.yaml may overwrite General settings in config.yaml

pdxlocations, 2025-02-19:
In my testing, femto_mac_address.yaml may cause defaults to be used for the General setting MaxNodes, and possibly others in config.yaml

Setting config.yaml to (note the db size):
```
General:
  MaxNodes: 500
  MaxMessageQueue: 100
  ConfigDirectory: /etc/meshtasticd/config.d/
#  MACAddress: AA:BB:CC:DD:EE:FF
#  MACAddressSource: eth0
```

and config.d/femto_mac_address.yaml to:
```
General:
  MACAddressSource: eth0
```

results in a max nodedb of the default 200.
```
Feb 18 22:55:30 femtofox meshtasticd[2519]: INFO  | 06:55:29 1800 [Router] Node database full with 200 nodes and 4294967295 bytes free. Erasing oldest entry
```
Removing the later file increases the db over 200.
```
Feb 18 23:18:51 femtofox meshtasticd[350]: INFO  | 07:18:43 763 [Router] Adding node to database with 212 nodes and 4294967295 bytes free!
```

noon92, 2025-02-19:
Hmmm, I guess adding "General" replaces everything in that category. I did not know that.

noon92, 2025-02-19:
I think this may be a meshtastic bug

pdxlocations, 2025-02-19:
> I think this may be a meshtastic bug

Clicking around I'm finding that yaml must use unique keys, or else previous declarations will be overwritten.
The firmware reads the "General" key and I believe at this point it only contains the mac assignment, so the default (200) is used
https://github.com/meshtastic/firmware/blob/191ca8ce124d9adbc9843ec5de0a3edf3f2c0af9/src/platform/portduino/PortduinoGlue.cpp#L535-L536

pdxlocations, 2025-02-19:
As this does not appear to be intended behavior, I am logging as a Meshtastic firmware issue: 
https://github.com/meshtastic/firmware/issues/6102