femtofox/femtofox #375: [Feature Request] Add tun/tap module to foxbuntu

jake-b, 2025-03-02:
I would like the tun/tap kernel module added to foxbuntu if possible. :)

You already added it, as I understand it, but you asked me to open an issue, so I did. :)

jake-b, 2025-03-18:
Hey can you just sanity check this one. I just installed the latest release and I did not see the tun module. 
I was able to compile my own .ko, but I found that while the tun driver was in menuconfig as <M> for module, the outer "Network core driver support" wasn't enabled.  I'm not sure if that was what was wrong.

NomDeTom, 2025-03-18:
Reopened by request

mikecarper, 2025-04-12:
[image]
Also looking for tun for tailscale vpn.

noon92, 2025-04-21:
Closed by #404

closed 2025-04-21 by noon92