femtofox/femtofox #397: Reading Battery Voltage

dswinton, 2025-03-30:
Hi Guys!

I'm testing out a couple of nodes built on Femtofox's, and they're absolutely awesome so far!  Thank you to everyone involved!

One thing I'm a bit stuck on though -- I've noted that I don't get a battery voltage reading.  I'm planning on deploying these remotely on solar, so having battery voltage readings would be a great help.

Is this something that should be possible to do on the hardware, or will I need to do something like add an INA219 over I2C or similar?

Thanks in advance!

Cheers

noon92, 2025-03-30:
Hi! This has absolutely been an issue - an i2c sensor would be necessary, but I believe meshtastic native still doesn't support power telemetry in any way shape or form. I recommend asking in #native on the meshtastic discord - perhaps @jonathanbennet.

dswinton, 2025-03-31:
Thanks @noon92 !

Since there's no hardware for voltage reading, i'll hook up an INA219.  That will solve my problem!

I'll raise the question over on Discord regarding the getting the info into meshtaticd.  If there's no solution there I'll put together a workaround with something like a bot that periodically messages the voltage or similar.

If there's a V2 of the board planned, I'd love to ask for voltage reading circuity to be a feature!  Either presented as an I2C sensor, or even just available somewhere in Linux (such as like a Laptop battery or something) would be awesome!

NomDeTom, 2025-04-03:
@dswinton Have you tried reading the analog voltage level on SARADC0 or 1, i.e. pins 4c0 and 4c1? I was looking to add a 3:1 voltage divider, so that the 5Vin can be read directly, and used to monitor an attached battery.

mikecarper, 2025-04-21:
https://github.com/SpudGunMan/meshing-around/blob/main/script/send-environment-metrics.py has the tooling to send out the info on the mesh

https://github.com/mikecarper/MeshtasticInstallScripts/blob/main/power.py has will read the INA219 on a pi zero

NomDeTom, 2025-04-23:
@dswinton 
> Thanks for the scripts [@mikecarper](https://github.com/mikecarper) - much appreciated!
> 
> I've also raised an issue under Meshtastic here regarding getting power telemetry on Meshtasticd: [meshtastic/firmware#6577](https://github.com/meshtastic/firmware/issues/6577)

There's a new library to handle ADC voltage reading and broadcast, which is apparently ready for testing: https://github.com/pdxlocations/mfoxadc