femtofox/femtofox #398: RAK / USB Periodically Dropping Out

dswinton, 2025-04-03:
Hi Guys,

This one is a strange issue I'm having trouble getting to the bottom of...  I'm hoping you can help!

I'm trying to run a RAK4361 as a second radio over USB. 
Initially it works, with the RAK showing up as /dev/ttyACMx and I'm able to configure and use it.

Periodically, the serial device will just completely disappear.  The timing isn't consistent, but its typically after a few hours.

Rebooting the RAK, reconnecting the USB or rebooting Linux on the Femtofox don't resolve the issue.  The only thing I've found to solve it is to disconnect the battery and cold boot the Femtofox.

I've tested with both the RAK powered over the USB port, as well as with power connected from battery, and it still happens either way.

I've tried 2 different complete builds (with a different Femtofox, RAK, USB cable, battery) and the same thing happens with both.

The RAK boards are both confirmed to be fine..  If I connect them to my Mac they show up as /dev/tty.* as normal.

I've also checked the USB port mode configuration with luckfox config and it's definitely set correctly.

lsusb output...

```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            5.10
  iManufacturer           3 Linux 5.10.160 xhci-hcd
  iProduct                2 xHCI Host Controller
  iSerial                 1 xhci-hcd.0.auto
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
        bMaxBurst               0
Hub Descriptor:
  bLength              12
  bDescriptorType      42
  nNbrPorts             0
  wHubCharacteristic 0x000a
    No power switching (usb 1.0)
    Per-port overcurrent protection
  bPwrOn2PwrGood       50 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  bHubDecLat          0.0 micro seconds
  wHubDelay             0 nano seconds
  DeviceRemovable    0x00
 Hub Port Status:
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x000f
  bNumDeviceCaps          1
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x02
      Latency Tolerance Messages (LTM) Supported
    wSpeedsSupported   0x0008
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   3
      Lowest fully-functional device speed is SuperSpeed (5Gbps)
    bU1DevExitLat           0 micro seconds
    bU2DevExitLat           0 micro seconds
Device Status:     0x0001
  Self Powered

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0002 2.0 root hub
  bcdDevice            5.10
  iManufacturer           3 Linux 5.10.160 xhci-hcd
  iProduct                2 xHCI Host Controller
  iSerial                 1 xhci-hcd.0.auto
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0019
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x000a
    No power switching (usb 1.0)
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0100 power
Device Status:     0x0001
  Self Powered
```

Would appreciate any pointers you can provide on how I could solve this!

dswinton, 2025-04-05:
A bit more troubleshooting info...

```
root@femtofox:~# cat /sys/devices/platform/ff3e0000.usb2-phy/otg_mode
host
```
Confirmed in host mode

Nothing appears in dmesg when disconnecting/reconnecting USB devices.

lsusb just shows the 2 busses...
```
root@femtofox:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

On further testing it seems to be a USB problem in general... not just affecting the RAK boards.  No USB devices seem to be recognised.

I did a full rebuild with a new luckfox, re-flashed SD card, new USB cable and new RAK board again.  This time around I seem to get no USB at all from the beginning.

I did notice something in dmesg... im not sure if this is expected?
```
[    0.191839] hub 2-0:1.0: config failed, hub doesn't have any ports! (err -19)
```

Here's the full USB section of dmesg during bootup...
```
[    0.188673] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    0.188699] ehci-pci: EHCI PCI platform driver
[    0.189194] phy phy-ff3e0000.usb2-phy.0: illegal mode
[    0.189219] xhci-hcd xhci-hcd.0.auto: xHCI Host Controller
[    0.189254] xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 1
[    0.189457] xhci-hcd xhci-hcd.0.auto: hcc params 0x0220fe64 hci version 0x110 quirks 0x0000000022010010
[    0.189510] xhci-hcd xhci-hcd.0.auto: irq 64, io mem 0xffb00000
[    0.190488] hub 1-0:1.0: USB hub found
[    0.190578] hub 1-0:1.0: 1 port detected
[    0.190966] xhci-hcd xhci-hcd.0.auto: xHCI Host Controller
[    0.190998] xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 2
[    0.191021] xhci-hcd xhci-hcd.0.auto: Host supports USB 3.0 SuperSpeed
[    0.191103] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[    0.191767] hub 2-0:1.0: USB hub found
[    0.191839] hub 2-0:1.0: config failed, hub doesn't have any ports! (err -19)
[    0.192319] usbcore: registered new interface driver cdc_acm
[    0.192341] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[    0.192410] usbcore: registered new interface driver cdc_wdm
[    0.192496] usbcore: registered new interface driver usb-storage
[    0.192628] usbcore: registered new interface driver usbserial_generic
[    0.192662] usbserial: USB Serial support registered for generic
[    0.192700] usbcore: registered new interface driver ch341
[    0.192726] usbserial: USB Serial support registered for ch341-uart
[    0.192790] usbcore: registered new interface driver usb_ch343
[    0.192798] ch343: USB serial driver for ch342/ch343/ch344/ch347/ch9101/ch9102/ch9103/ch9104, etc.
[    0.192804] ch343: V1.5 On 2022.12
[    0.192851] usbcore: registered new interface driver cp210x
[    0.192879] usbserial: USB Serial support registered for cp210x
[    0.192938] usbcore: registered new interface driver usb_serial_simple
[    0.192965] usbserial: USB Serial support registered for carelink
[    0.192987] usbserial: USB Serial support registered for zio
[    0.193009] usbserial: USB Serial support registered for funsoft
[    0.193030] usbserial: USB Serial support registered for flashloader
[    0.193053] usbserial: USB Serial support registered for google
[    0.193075] usbserial: USB Serial support registered for libtransistor
[    0.193096] usbserial: USB Serial support registered for vivopay
[    0.193117] usbserial: USB Serial support registered for moto_modem
[    0.193147] usbserial: USB Serial support registered for motorola_tetra
[    0.193169] usbserial: USB Serial support registered for nokia
[    0.193194] usbserial: USB Serial support registered for novatel_gps
[    0.193217] usbserial: USB Serial support registered for hp4x
[    0.193248] usbserial: USB Serial support registered for suunto
[    0.193270] usbserial: USB Serial support registered for siemens_mpi
[    0.193453] gadgetfs: USB Gadget filesystem, version 24 Aug 2004
[    0.193470] udc-core: couldn't find an available UDC - added [g_mass_storage] to list of pending drivers
```

I'm still at a bit of a loss about how to fix this.  I suspect it's some kind of hardware issue, but remain hopeful it might just be a setting somewhere 😅

If anyone has a RAK board successfully working for more than a day or so connected to a Femtofox I'd appreciate any insight or pointers, please!

One note is that I'm using the community edition board, so the USB port I'm using is the one on the Luckfox itself.  There's no USB port on my Femtofox's (which might explain that error at post, maybe?).

I noted that the RAK is listed as a supported device over both USB and UART.  Was that tested using the Luckfox port, or the Femtofox one?

I'm going to try running it over UART next, but I'm not sure if my intended use case will work over UART (out of band network TUN over mesh).

noon92, 2025-04-05:
I'm wondering if this is a hardware issue, like you suspect. @Ruledo can you try to replicate this?

dswinton, 2025-04-06:
@noon92 @Ruledo I think I found the cause!

I'm powering the Femtofox from a battery.  It seems that when the battery voltage drops to low (not sure on the exact threshold yet) then the USB becomes unreliable.  At a certain point below that, the USB stops working completely.

I was able to replicate the issue with 2 completely separate setups -- different Femtofox, luckfox and RAK... and the fault was following the (partially discharged) battery.

If I run the system from a battery fully charged to 4.2 volts... everything works perfectly.
Also to test further I tried on a bench PSU fixed to 5 volts... also all working there.

I guess to solve this I need to make sure the voltage stays high enough, so I will put a boost regulator in between the battery and the Femtofox's power port to keep a steady 5 volts.  Not ideal as it will have an efficiency cost, but it should fix it.

My guess is that this issue would likely affect others... so if anyone else can replicate this issue it might be something we should note in the docs somewhere to save others having the same struggle I have.

noon92, 2025-04-06:
Ahhhh this makes sense!!! I was wracking my brain on this, but it just didn't track. So femtofox works with 3.3-5v, but anything below 5v starts losing tx power. You definitely need a board to provide steady voltage.

noon92, 2025-04-06:
I'll leave this issue open until you confirm that was the issue - and if so, I'll update the docs to include this info and close it. Thank you for tracking this one down!

dswinton, 2025-04-07:
@noon92 Yep -- confirming its been running perfectly since putting a boost regulator to 5V in.

Definitely worth noting in the docs - something to the effect of "If you're using USB,  you need to deliver 5V to the board or weird things will happen 🤣 "

noon92, 2025-04-25:
I'm closing this

closed 2025-04-25 by noon92