femtofox/femtofox #402: Add tailscale to install menu

bbrodka2, 2025-04-16:
While we can probably manually install tailscale, it would be a nice addition to the femtofox software suite.
Tailscale makes creating software-defined networks easy: securely connecting users, services, and devices.

Create a VPN to manage all femtofox nodes no matter where they are from one central point

Its free forever for 3 users and 100 devices
more info;
https://tailscale.com/

dswinton, 2025-04-17:
@bbrodka2 you might want to merge over to this issue:  https://github.com/femtofox/femtofox/issues/375

Installing Tailscale itself is straightforward but you will need the TUN module for it to work, which I think is currently a manual compile.

noon92, 2025-04-25:
Please talk to me on discord (username noon) to discuss further.

mikecarper, 2025-05-14:
I was able to get this working.  

Enable TUN  
```
sudo femto-config
5 (Utilities & Tools) -> 3 (Kernel module manager) -> Yes -> tun -> Enable
```

Install Tailscale  
```
curl -fsSL https://tailscale.com/install.sh | sh
```

Set Tailscale netfilter mode to off  
```
sudo systemctl stop tailscaled
sudo systemctl edit tailscaled
```
Have the override.conf file look like this  
```
### Editing /etc/systemd/system/tailscaled.service.d/override.conf
### Anything between here and the comment below will become the new contents of the file

[Service]
Environment="TS_NETFILTER_MODE=off"
ExecStart=
ExecStart=/usr/sbin/tailscaled --state=/var/lib/tailscale/tailscaled.state --socket=/run/tailscale/tailscaled.sock --tun=userspace-networking

### Lines below this comment will be discarded
```

Reload and startup Tailscale  
```
sudo systemctl daemon-reload
sudo systemctl restart tailscaled
sudo tailscale up
```

Xaositek, 2025-05-15:
Tested this on my rebuilt FemtoFox - this worked perfectly! Great job @mikecarper

MIXTERek, 2026-03-22:
Could you please share an image containing this change? Unfortunately, I keep getting errors while building