femtofox/femtofox #415: Can not see messages on Femtofox.

Majikterror, 2025-05-21:
Can not see direct messages, public channel, or private channel messages. But all nodes are getting Ack from Femtofox when direct messaging. Any ideas? Start over?

Majikterror, 2025-08-24:
Will give it go. Seems like a pretty simple solution, lol.

Brabouk, 2025-09-26:
> Will give it go. Seems like a pretty simple solution, lol.

Hi, did this help solve the problem? I still remember that this problem occurs every time I update meshtastic. That's why I always update and upgrade the system after updating MT.

Majikterror, 2025-09-26:
Still haven't touched it. I will eventually. The fact that I cant connect
to this node with bluetooth is extremely limiting for my use case. I guess
I should of researched more before buying it.

itxpress, 2025-09-30:
I think this is an issue with meshtasticd in general, but it seems worse with the femtofox. I have a Femtofox and a Pi Zero 2 with a MeshAdv Pi Hat. I only can get messages when I'm actively connected to the node. it will NOT store received messages and send them to the app when  you connect like most nrf/ESP nodes do. I see this with both nodes. That said, I put 2.7.x on my femtofox (which has a new improved webgui) and for a while it would occasionally load a handful of previous messages. But not always, then it stopped doing it. For my Pi node, the Android app won't load anything, but MeshSense loads everything including past received messages just like a normal node. It's very weird. The Android App has been messed up when connecting to network based nodes for a while now. So it may just be that. I'm wiping my fox and reinstalling fresh to see if I can get it to act like a normal mode in regard to messages. I'm glad to see the recommendation to update the system overall with apt-get. I was worried that might be breaking things compared to the 1.0.0 stock image.